FreeCAD/FreeCAD #24336: Assembly: recompute issue related to expression in advanced offset

Roy-043, 2025-09-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the attached assembly expressions are used in advanced offsets. One of the joint only gets recomputed if you select it individually.

1. Unpack the ZIP file in a single folder.
2. Open tangent_assy_roy2.FCStd.
3. Change the Distance property of the Offset joint from 20mm to 50mm.
4. Recompute.
5. Result: only one of the plates is in the correct position.
6. Mark the whole file for recomputation.
7. Recompute.
8. Result: nothing has changed.
9. Manually recompute the Edge_Tangency002 joint.
10. Result: 2nd plate is in the correct position.

[tangent_assy_roy2.zip](https://github.com/user-attachments/files/22617230/tangent_assy_roy2.zip)

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Krypto31337, 2025-10-02:
Confirmed.  When adjusting Distance Setting of Offset Joint as described, user needs to Right Click on Edge_Tangency002 and select "Mark to Recompute" and then Recompute document for the 2nd plate to move to the correct position.  Recompute document by itself will move one plate but not both.

```
OS: Linux Mint 22.2 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.1104/1
Installed mods: 
  * Telemetry 1.0.2
  * OpenTheme 2025.5.20
  * freecad.gears 1.3.0
  * sheetmetal 0.7.55
  * fasteners 0.5.42
```

PaddleStroke, 2025-11-19:
- it does not appear to be specific to advanced offset. Creating a 'test' property to this Joint001.Distance behaves the same.

PaddleStroke, 2025-11-19:
Oh well this issue is turning in the journey down the rabbit hole of assembly recompute mess.