FreeCAD/FreeCAD #24339: pre-commit.ci fails with timeout

FlachyJoe, 2025-09-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[image]

### Full version info

```shell
since 2025-09-30
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

chennes, 2025-10-01:
I wrote to the pre-commit.ci team to see what our options are here: as far as I know there are no logs directly accessible to us, so I can't see if something changed on our end that broke this, or if we were always just a little under the three minute limit and are not just a little over. I don't really know why it takes so long to run, on my systems it runs in just a few seconds.

chennes, 2025-10-01:
Here's some detail back from the pre-commit team:

----

I usually try and debug this by using `--verbose` to get timing information out

in this case it appears `black` is taking ~most of the time? perhaps on some files that are quite large or it is unhappy with

for instance, this file alone seems to take about 50% of the execution time on my machine:

```
$ pre-commit run --verbose --files src/Mod/Fem/femexamples/meshes/mesh_beamsimple_tetra10.py
...
```

here's some of the files black might be struggling with:

```
$ git ls-files -- '*.py' | xargs wc -l | grep -E '[0-9]{5}'
   16985 src/Mod/Fem/femexamples/meshes/mesh_beamsimple_tetra10.py
   18197 src/Mod/Fem/femexamples/meshes/mesh_buckling_ibeam_tria6.py
   10931 src/Mod/Fem/femexamples/meshes/mesh_constraint_centrif_tetra10.py
   18522 src/Mod/Fem/femexamples/meshes/mesh_constraint_tie_tetra10.py
   22986 src/Mod/Fem/femexamples/meshes/mesh_contact_tube_tube_tria3.py
   14374 src/Mod/Fem/femexamples/meshes/mesh_electricforce_elmer_nongui6_tetra10.py
   19953 src/Mod/Fem/femexamples/meshes/mesh_multibodybeam_tetra10.py
   19766 src/Mod/Fem/femexamples/meshes/mesh_platewithhole_tetra10.py
   41992 src/Mod/Import/App/SCL_output/ap203_configuration_controlled_3d_design_of_mechanical_parts_and_assemblies_mim_lf.py
   44542 src/Mod/Import/App/SCL_output/automotive_design.py
   18442 src/Mod/Import/App/SCL_output/config_control_design.py
   37012 src/Mod/Import/App/SCL_output/ifc2x3.py
   43883 src/Mod/Import/App/SCL_output/ifc4.py
```

you may also be able to switch to the "precompiled" black which speeds things up a bit?

```diff
$ git diff
diff --git a/.pre-commit-config.yaml b/.pre-commit-config.yaml
index 84612dc..85f418d 100644
--- a/.pre-commit-config.yaml
+++ b/.pre-commit-config.yaml
@@ -60,8 +60,8 @@ repos:
     -   id: check-yaml
     -   id: check-added-large-files
     -   id: mixed-line-ending
--   repo: https://github.com/psf/black
-    rev: af0ba72a73598c76189d6dd1b21d8532255d5942  # frozen: 25.9.0
+-   repo: https://github.com/psf/black-pre-commit-mirror
+    rev: 25.9.0
     hooks:
         -   id: black
             args: ['--line-length', '100']
```

(about a ~30% improvement on my machine -- obviously you'd want to refreeze it since that seems your repo's preference)