FreeCAD/FreeCAD #24342: PartDesign: Pipe does not work fine when the Sketch is inclined - Operation and Preview are also not getting the same conclusion.

leoheck, 2025-09-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I have these 2 sketches; the path is not horizontal, it is inclined 0.5 degrees.

[image]

[image]

When doing the PD:Pipe the preview works with the Standard, Fixed, and Frenet modes.
But when committing the feature, Fixed does not work.

[image]

Produced features are not what the preview showed.

[image]

This is what it should look like (here using Frenet). Standard looks the same here. Are they the same thing?

[image]

The feature should move on an inclined Sketch. 
Everytime I use sketch with an angle (on x axis) features missbehave.

<hr/>

Now, sometimes we have a preview that fails, but the operation succeeds. 

[image]

[image]

[image]

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Assembly4.1 0.60.2-0.1
  * Cables 0.2.1
  * CurvesWB 0.6.64
  * FreeCAD-render 2024.12.15
  * Assembly4 0.60.2
  * sheetmetal 0.7.55
  * DynamicData 2.75.0
  * Curves 0.6.70
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * Assembly4.1_mod 0.60.2-0.1
  * OpenTheme 2025.5.20
  * fasteners 0.5.42
  * Silk 0.2.0
  * osh-autodoc-workbench 0.2.3
  * CurvedShapes 1.0.13
  * freecad-povray-render
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.7.6
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-09-30:
This looks like an orientation thing.
A couple of days ago, I noticed that the orientation of lines was important.

Now trying to overcome this issue with the pipe, and doing a loft instead with these 2 sketches.

[image]

The produced output looks switching the orientation of things.
This may be something new since such results have never happened to me.

This is the preview of this subtractive loft. Some X's appear, showing the geometry switches direction. 

[image]

leoheck, 2025-10-01:
I could find a solution that works instead of having to use a rotated Sketch to create this unleveled path.

I made a SubShape binder that reproduces the inclined perimeter of the top.
In this scenario, both the Additive and the Subtractive Loft worked. 

[unleveled_path.FCStd.zip](https://github.com/user-attachments/files/22646667/unleveled_path.FCStd.zip)

[image]

kaiwas, 2025-10-03:
Maybe I should use a binder? I don't know how it works, but everything looks pretty even.

https://github.com/user-attachments/assets/7496853e-9313-467f-ac13-78eed322e503